wokwi/wokwi-features #794: A project can be saved only once, when not logged in.

Koepel, 2024-05-29:
This problem was mentioned on the Discord channel.

When **not** logged in, then it is possible to save a project, but only once.
The first time the project is saved, a unique url to the new project is created. That new url is good and valid.

Without leaving that page, and after editing the code, it is not possible to save it any more.
Both Firefox and Chrome present an error message: "Project saving failed! Your changes have not been saved."

When I try to open that link in an other browser, then a box is shown with: "This project is owned by a different user. You can save a copy of this project in your account.". That is okay. I can save a copy.
But then the same problem occurs, I can not save that copy a second time.

urish, 2024-05-31:
Issue reproduced

urish, 2024-05-31:
Fix is up - can you please try now and report?

Koepel, 2024-05-31:
Extensively tested by changing things in the code, diagram, title and between multiple browsers.
Everything works as expected.

urish, 2024-06-01:
Thank you so much!

closed 2024-06-01 by urish

alto777, 2025-01-01:
I am having a similar problem lately when posing as an anonymous user. Dunno if it's the same exactly, but certainly at some point saving an anonymous project at the same workstation it was created on does not work.

I can't say it fails if I never leave the actual window where I first saved.