wokwi/wokwi-features #798: Neopixels: glitchy after the first few LEDs with ESP32-IDF+SPI

scoates, 2024-06-07:
Hello. Starting out by saying that wokwi is amazing. There's so much here, and I hope you've proud of the work that's gone into this. I was happy to join the club and send you some dollars.

**Describe the bug**

There seems to be a bug in the simulated `wokwi-neopixel-matrix` where it glitches out after the first 6 (or so) ws2812s in the chain.

[image]

(This is running on the result of `idf.py flash` from the same binary that's installed in the wokwi simulator in the above screenshot.)

I've tried this on real hardware, and it works as I'd expect (more than 6 LEDs are able to be addreesed). 

https://github.com/wokwi/wokwi-features/assets/71983/54a41cda-ba4e-4720-8e01-7504bec7eb28

This is on ESP32-C6 with IDF and lighting the pixels via SPI (you can see this in the code).

I've also tried this with stringing a bunch of independent neopixels together (in wokwi; the simulator), and it also glitches out, so it's not exactly the panel that is failing; it's probably the SPI timings? I didn't manage to get it to work at all with RMT instead of SPI, so I'm not sure if it's directly tied to SPI emulation. I do know that SPI works on the real hardware.

I don't really know enough about SPI and timings to do much with the logic analyzer, so I don't have that to share, but someone with the right knowledge can attach that to the project easily, I'd think.

**To Reproduce**
I have a [repository over here](https://github.com/scoates/wokwi_broken_panel) that illustrates the problem. Run this code in the simulator and it fails. There's a [UF2 here](https://files.scoat.es/VsvF8JRfS2.bin) which can be uploaded into [this example sim](https://wokwi.com/projects/400000038647312385) and it shows the problem (the default code only shows the hello message, use F1 + Upload the downloaded uf2.bin).

**Expected behavior**
I expect the 9 LEDs to light up blue both on the real-world matrix and on the simulated one. But the Simulated one glitches after 6. I limited my test to 9 but it fails after 6, here, even if I try to address all 256. Some are unlit, many are white; many are the wrong colour. It really looks like a timing issue which I experienced on other platforms.

**Environment:**
 - OS: not sure it's relevant, but MacOS, M2 MBP, ESP32 C6 + IDF.
 - Browser: happens both in Firefox and in the VSCode Extension.

scoates, 2024-06-08:
FWIW, I tried this on the simulator only with the `c3` and `s3` boards (the `h2` board doesn't seem to have the spi capability in IDF), and it failed (glitchy) with even fewer blue neopixels.

urish, 2024-06-14:
Thanks for the super detailed report Sean, it's really helpful.  Also thanks for all the kind words! ❤️ 

I was able to reproduce the issue, and am now looking into it.

scoates, 2024-06-17:
Thanks for taking a look. Let me know if I can help in any way.

urish, 2024-06-17:
Update: I believe I found the cause - it's a combination of too tight timing margins in the WS2812 decoder together with inaccurate timing of the SPI transmitter. Working on a fix.

urish, 2024-06-18:
Pushed a fix - can you please test and report?

scoates, 2024-06-18:
Yes! All 256; no glitches. Thank you!

[image]

closed 2024-06-18 by urish