wokwi/wokwi-features #799: SH1107 display not working

Mer08, 2024-06-07:
**Describe the bug**
Code works, display doesn't. Serial console works fine, that's how I managed to tell if it was running. It was working just fine at home, idk what happened

**To Reproduce**
Load up my project, hit run. https://wokwi.com/projects/399869981609543681

**Expected behavior**
Text is printed and screen is cleared.

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Browser Chrome AND Firefox
 - Version (Firefox 114.0.1) (Chrome 103.0.5060.53)

**Additional context**
When loading the project I get these errors on both browsers.
```
Uncaught Error: Minified React error #418; visit https://reactjs.org/docs/error-decoder.html?invariant=418 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at lg (framework-128cfde67f3541f2.js:9:46339)
    at i (framework-128cfde67f3541f2.js:9:121283)
    at oD (framework-128cfde67f3541f2.js:9:99073)
    at framework-128cfde67f3541f2.js:9:98940
    at oO (framework-128cfde67f3541f2.js:9:98947)
    at ox (framework-128cfde67f3541f2.js:9:93986)
    at x (framework-128cfde67f3541f2.js:33:1364)
    at MessagePort.T (framework-128cfde67f3541f2.js:33:1894)
framework-128cfde67f3541f2.js:9 Uncaught Error: Minified React error #418; visit https://reactjs.org/docs/error-decoder.html?invariant=418 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at lg (framework-128cfde67f3541f2.js:9:46339)
    at i (framework-128cfde67f3541f2.js:9:121106)
    at oD (framework-128cfde67f3541f2.js:9:99073)
    at framework-128cfde67f3541f2.js:9:98940
    at oO (framework-128cfde67f3541f2.js:9:98947)
    at ox (framework-128cfde67f3541f2.js:9:93986)
    at x (framework-128cfde67f3541f2.js:33:1364)
    at MessagePort.T (framework-128cfde67f3541f2.js:33:1894)
2framework-128cfde67f3541f2.js:9 Uncaught Error: Minified React error #418; visit https://reactjs.org/docs/error-decoder.html?invariant=418 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at lg (framework-128cfde67f3541f2.js:9:46339)
    at i (framework-128cfde67f3541f2.js:9:121283)
    at oD (framework-128cfde67f3541f2.js:9:99073)
    at framework-128cfde67f3541f2.js:9:98940
    at oO (framework-128cfde67f3541f2.js:9:98947)
    at ox (framework-128cfde67f3541f2.js:9:93986)
    at x (framework-128cfde67f3541f2.js:33:1364)
    at MessagePort.T (framework-128cfde67f3541f2.js:33:1894)
6framework-128cfde67f3541f2.js:9 Uncaught Error: Minified React error #418; visit https://reactjs.org/docs/error-decoder.html?invariant=418 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at lg (framework-128cfde67f3541f2.js:9:46339)
    at i (framework-128cfde67f3541f2.js:9:121106)
    at oD (framework-128cfde67f3541f2.js:9:99073)
    at framework-128cfde67f3541f2.js:9:98940
    at oO (framework-128cfde67f3541f2.js:9:98947)
    at ox (framework-128cfde67f3541f2.js:9:93986)
    at x (framework-128cfde67f3541f2.js:33:1364)
    at MessagePort.T (framework-128cfde67f3541f2.js:33:1894)
framework-128cfde67f3541f2.js:9 Uncaught Error: Minified React error #423; visit https://reactjs.org/docs/error-decoder.html?invariant=423 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at i (framework-128cfde67f3541f2.js:9:120775)
    at oD (framework-128cfde67f3541f2.js:9:99073)
    at framework-128cfde67f3541f2.js:9:98940
    at oO (framework-128cfde67f3541f2.js:9:98947)
    at oE (framework-128cfde67f3541f2.js:9:95699)
    at ox (framework-128cfde67f3541f2.js:9:94254)
    at x (framework-128cfde67f3541f2.js:33:1364)
    at MessagePort.T (framework-128cfde67f3541f2.js:33:1894)
```

And when running project (only Firefox I believe):
```
Failed to paste diagram parts: SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data
    NextJS 26
        to
        m
        uU
        oV
        o
        x
        T
        60053
        p
        63840
        p
        64448
        p
        73935
        p
        20745
        p
        38060
        p
        1775
        p
        <anonymous>
        O
        <anonymous>
        u
        <anonymous>
```

Mer08, 2024-06-07:
Tested on Opera GX LVL 5 109.0.5097.108 and it still doesn't work.
Error log:
```
Uncaught Error: Minified React error #418; visit https://reactjs.org/docs/error-decoder.html?invariant=418 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at lg (framework-128cfde67f3541f2.js:9:46459)
    at i (framework-128cfde67f3541f2.js:9:121106)
    at oD (framework-128cfde67f3541f2.js:9:99073)
    at framework-128cfde67f3541f2.js:9:98940
    at oO (framework-128cfde67f3541f2.js:9:98947)
    at ox (framework-128cfde67f3541f2.js:9:93986)
    at x (framework-128cfde67f3541f2.js:33:1364)
    at MessagePort.T (framework-128cfde67f3541f2.js:33:1894)
framework-128cfde67f3541f2.js:9 Uncaught Error: Minified React error #423; visit https://reactjs.org/docs/error-decoder.html?invariant=423 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at i (framework-128cfde67f3541f2.js:9:120775)
    at oD (framework-128cfde67f3541f2.js:9:99073)
    at framework-128cfde67f3541f2.js:9:98940
    at oO (framework-128cfde67f3541f2.js:9:98947)
    at oE (framework-128cfde67f3541f2.js:9:95699)
    at ox (framework-128cfde67f3541f2.js:9:94254)
    at x (framework-128cfde67f3541f2.js:33:1364)
    at MessagePort.T (framework-128cfde67f3541f2.js:33:1894)
```
Please, look into this. it was working just fine yesterday. in fact, Opera GX caches websites when my browser is closed, so when I opened it when I came home, it hadn't changed at all and it's working! It's def a problem on your side, because the cached website works fine, and when I update the code and run it, it works OK!!!

urish, 2024-06-08:
Hi, thanks for reporting. I looked into this, but it seems like the text displays on the screen:

[image]

Mer08, 2024-06-08:
Oh great, thanks! Should this issue be closed?

closed 2024-10-14 by urish