wokwi/wokwi-features #801: Stepper motor with A4988 makes sometimes -2 steps at startup, depending on the order of parts in diagram.json

Koepel, 2024-06-11:
Only a A4988 and a stepper motor are needed. The four wires to the stepper motor must be connected and the SLEEP must be connected to RESET.

When the A4988 is before the stepper motor in diagram.json, then the stepper motor stays at 0 steps when the simulation is started.
When the stepper motor is first, then the motor makes -2 steps.

0 steps: https://wokwi.com/projects/400378484824646657
-2 steps: https://wokwi.com/projects/400378469835249665

urish, 2024-06-11:
Good catch!

urish, 2024-06-11:
Pushed a fix - can you please test again?

Koepel, 2024-06-11:
Everything is okay.

Tested with Firefox and Chromium and a very long and slow diagram.json file with the A4988 at the end. No problem.

urish, 2024-06-12:
Thanks for testing and reporting back!

closed 2024-06-12 by urish