wokwi/wokwi-features #805: Error when using IDF cpp (not c) code on esp32

AbdelwahedTahri, 2024-06-15:
**Describe the bug**
The bug happens when you use an IDF cpp code with esp32.

# if you don't use idf template
- Before i was able to run a code containing class declaration and definition and a main_app() content without any problem and now i get this message -> "multiple definitions of app_main"
- i can understand that the reason is that the main_app() is declared also on main.cpp within arduino core library
- but i'm saying that before it was working, idk why but it was 

# if u use idf template
In this case you can't run cpp code (classes, templates, lambdas ...) because the compiling script uses the c compiler.
and adding extern "C" generate another error

**To Reproduce**
https://wokwi.com/projects/400791516965254145
https://wokwi.com/projects/350976545001570899

Simply you can't have a definition of `app_main()` within your source file

**Expected behavior**
what expected is that the program compile and run as it was before without telling me "multiple definitions of app_main"

**Environment (please complete the following information):**
 - Browser : firefox
 - Version idk

**Additional context**
Add any other context about the problem here.

OsvaldoTCF, 2024-09-29:
So, to have the same behavior as before using C++ files, IDF libraries, etc., try use the ".ino" extension - that is, compile for Arduino - rearrange loop() and setup() with the empty loop() and the setup() with a function that would be the app_main() of an IDF application.
Without the bulker configuration for idf:
"attrs": { }

Maybe it will help until there is a definitive solution consistent with the previous behavior:
- IDF, C++ with .cpp and .hpp - without Arduino

https://wokwi.com/projects/410303509952875521

OsvaldoTCF, 2024-10-01:
As expected, functions that require the IDF directly do not work. Or, at least I couldn't make it work.
This is the case with the esp_log.h.

The issue is that we **need IDF builder mode:
"attrs": {    "builder": "esp-idf"   }
and
C++ (main.cpp, *.hpp, etc)**