wokwi/wokwi-features #806: ESP32-S2 UART issues

lucasssvaz, 2024-06-17:
**Describe the bug**
The output of some ESP32-S2 tests consistently have wrong bytes/garbled text. It started happening this last day.

**To Reproduce**
Check CI output:
https://github.com/espressif/arduino-esp32/actions/runs/9540907794/job/26312774907

**Expected behavior**
Normal output like the other chips.

**Environment (please complete the following information):**
 - OS: ubuntu-latest

urish, 2024-06-17:
Issue reproduced - researching what causes it

urish, 2024-06-17:
Looks like it's related to wrong initial value for the PRE_DIV_CNT field in the SYSCLK_CONF conf register, causing UART baud rate to go totally crazy, initially:

[image]

 Working on a fix.

urish, 2024-06-17:
Fix is up - can you please test again now?

lucasssvaz, 2024-06-17:
@urish looks good. Everything working as expected. Thanks!

closed 2024-06-17 by lucasssvaz

urish, 2024-06-17:
Thanks for spotting this!