wokwi/wokwi-features #808: Cannot control some IOs of ESP32S3-dev board

martinberlin, 2024-06-20:
**Describe the bug**

I'm trying to make a simple array to set the referenced IOs as Output and at the same time turn them high. 
This seems to work for many GPIOs but others seem not to react, even after the array loop, trying to explicitly do it like this:

```C
  digitalWrite(41, HIGH);
  digitalWrite(48, HIGH);
```

I'm not sure this is a **bug** but I can certainly turn all GPIOs high in a physical board.

**To Reproduce**

https://wokwi.com/projects/401153784260252673 

**Expected behavior**

Leds being lit when the connected IO is set as output and turned on high.

urish, 2024-06-20:
Thanks for reporting, investigating.

Koepel, 2024-06-20:
It is hard to tell which pins can sometimes not be set as output, it seems to depend on other pins.
With my tests, pin 3 is the one that is most likely that does not work.

Koepel, 2024-06-20:
Setting pin 3 as output works, but not when all the pins are set as output: https://wokwi.com/projects/401202889224291329

Koepel, 2024-06-20:
My test was wrong.

Koepel, 2024-06-20:
I fell into the same trap, I was looking at the wrong pin numbers, sorry.

Changing the array for martinberlin to the actual pin numbers makes it work:
```
// Fast signals XCL,STV
int sigIO[5] = {5, 47, 48, 40, 41};
// 8 data out
int dataIO[8] = {6,7,15,16,17,18,8,3};
```

urish, 2024-06-20:
I agree this is confusing - I'll ask @AriellaE to look at how to make it less confusing.

martinberlin, 2024-06-20:
> Note that the label of the pin comes below the pin (I do find it a bit confusing too, but this is how the actual S3 devkit is marked)

Yes my bad that's 100% right. I will adapt the pins and update it, that is mostly the issue. 
Another thing that I find strange is that pins 41 & 47 seem to be high even without setting them (That's why I added a delay) But that is a normal thing is many micro-controllers. And is a good reason to double check and always make real tests before assigning pins that control mosfet's or other states, since they might toogle in the boot process. This happened to me controlling backlights and is always a bit of a struggle.

martinberlin, 2024-06-20:
I updated the title since this is solved and I was too lazy to make just another Issue to figure out the PSRAM issue. 
The fact is, even if I add the attributes for external RAM:

```json
{
  "parts": [
    { "type": "board-esp32-s3-devkitc-1", 
     "id": "esp", "top": -0.18, "left": -5.03, 
    "attrs": { "psramSize": "8", "flashSize": "4" }
}
```

Hi @urish and @Koepel 
A demo in the previous link to the project. Checking if PSRAM is recognized with:

```C
  if(psramInit()){
    Serial.println("\nPSRAM is correctly initialized");
    Serial.print("Total PSRAM:");
    Serial.println(ESP.getPsramSize());
    Serial.print("Free PSRAM: ");
    Serial.println(ESP.getFreePsram());
  }else{
    Serial.println("PSRAM not available");
  }
  ```

in a physical board using Arduino IDE and setting the Tools-> PSRAM to enabled it works. Doing the JSON setting with Wokwi it does not and that will really not let me simulate the environments and libraries I want, since I use big displays, and many components are highly dependant on external RAM to draw their display framebuffers.

urish, 2024-06-24:
Board graphics updated to avoid future confusion about the pin numbers.

closed 2024-06-24 by urish