wokwi/wokwi-features #810: ESP and Micropython error

gitwazza, 2024-06-24:
When I run any Micropython code for the ESP32 I get the following message. "The filesystem appears to be corrupted. If you had important data there, you may want to make a flash snapshot to try to recover it. Otherwise, perform factory reprogramming of MicroPython firmware (completely erase flash, followed by firmware programming)."**Describe the bug**
A clear and concise description of what the bug is. 

This is my project from 2 yrs ago https://wokwi.com/projects/334061696710607444 - that now produces the above error.

If I try and run a sample for the ESP32-S3 (https://wokwi.com/projects/new/micropython-esp32-s3) - the board LED goes on, but the 'print("Hello, ESP32-S3!")' statement doesn't produce any output in the terminal.

urish, 2024-06-24:
Thanks for reporting!

Pushed a fix, can you please check again?

gitwazza, 2024-06-24:
Hi Urish,
Thanks for the fix. It all appears to be working now.
Regards
Warren

closed 2024-06-24 by gitwazza