wokwi/wokwi-features #811: ESP32 Arduino - NVS stopped working

P-R-O-C-H-Y, 2024-06-24:
**Describe the bug**
In recent wokwi version the NVS stopped working for ESP32, ESP32S2, ESP32S3 and ESP32C3.

Link to failed CI run:
https://github.com/espressif/arduino-esp32/actions/runs/9641334301/job/26586787348

**To Reproduce**
Tests binaries can be downloaded from CI run in artifacts:
https://github.com/espressif/arduino-esp32/actions/runs/9641314348

**Expected behavior**
Test should pass as it works on real HW and worked before on Wokwi also.

**Environment (please complete the following information):**
Github CI: platform linux -- Python 3.12.4, pytest-8.2.2, pluggy-1.5.0

urish, 2024-06-24:
I believe this was the same issue as #810 and should now be fixed, can you please try again?

P-R-O-C-H-Y, 2024-06-24:
Still not working for me.

urish, 2024-06-24:
Yeah, the issue reproduces for the on the ESP32-C3. Looking into it.

urish, 2024-06-24:
Issue found, working on a fix.

urish, 2024-06-24:
Pushed a fix - can you please test once more?

P-R-O-C-H-Y, 2024-06-24:
> Pushed a fix - can you please test once more?

ESP32S2 still not working. 
But S3 and C3 fixed. 

https://github.com/espressif/arduino-esp32/actions/runs/9645751282/job/26600656491#step:11:188

urish, 2024-06-24:
Thanks! Issues found, working on a fix.

urish, 2024-06-24:
Pushed a fix. 

Also, in order to avoid future NVS regressions, I added the ESP-IDF nvs driver tests to our CI:

- Before the fix: https://github.com/wokwi/esp32-test-binaries/actions/runs/9652601127/job/26623081200 🔴 (S2/C6/H2/P4 failing)
- Now: https://github.com/wokwi/esp32-test-binaries/actions/runs/9652694870/job/26623386390 🟢

P-R-O-C-H-Y, 2024-06-25:
@urish Thank you for a quick fix! All tests passing now. Great job.

closed 2024-06-25 by P-R-O-C-H-Y