wokwi/wokwi-features #812: ESP32 SD Card doesn't work

TheVirtualFox, 2024-06-24:
**Describe the bug**
Exemple from SD.h library doesn't work, but it worked before.

https://wokwi.com/projects/323656763409695316

urish, 2024-06-24:
Thanks for reporting, issue reproduced. Will investigate!

urish, 2024-06-25:
Issue found - seems like the SPI MISO pin does not get configured correctly. Working on a fix.

urish, 2024-06-25:
Fixed!

closed 2024-06-25 by urish

TheVirtualFox, 2024-06-25:
thank you. can i get link to commit?

urish, 2024-06-25:
You're welcome. The ESP32 simulation core is not open source, but if you are curious, this is the diff of the fix:

```diff
diff --git a/src/peripherals/gpio.ts b/src/peripherals/gpio.ts
index 8785a6d..d79fcb1 100644
--- a/src/peripherals/gpio.ts
+++ b/src/peripherals/gpio.ts
@@ -321,14 +321,16 @@ export class IOPin {
     // TODO: LP override?
     const signals = new Set<IPinSignalInfo | string>();
     let signalInfo = this.muxConfig?.fn[this.muxFunction];
-    if (signalInfo === PinSignal.GPIO && this.matrixEnable) {
+    if (signalInfo === PinSignal.GPIO) {
       const signalOut = this.gpio.cpu.gpioMatrix.pinSignalOutInfo(this.id);
-      if (signalOut) {
+      if (signalOut && this.matrixEnable) {
         signals.add(signalOut);
       }
       for (const inSignal of this.gpio.cpu.gpioMatrix.pinSignalInInfo(this.id)) {
         signals.add(inSignal);
       }
+    } else if (signalInfo) {
+      signals.add(signalInfo);
     }
     return signals;
   }
```