wokwi/wokwi-features #814: [esp32c3]: Interrupt stopped working

SergioGasquez, 2024-06-25:
**Describe the bug**
The following Wokwi project:  https://wokwi.com/projects/382723722184136705?build-cache=disable has an interrupt attached to the button which should print a message when the button is pressed, for some reason that stopped working. 

That example is check daily in CI https://github.com/esp-rs/no_std-training/actions/workflows/wokwi_projects.yml and it used to work.

Also, if I download the project ELF and flash it to a board, it works as expected.

**To Reproduce**
1. Run the Wokwi project
2. Press the button
3. Nothing happens

**Expected behavior**
1. Run the Wokwi project
2. Press the button
3. The interrupt prints a message

urish, 2024-06-25:
Thanks for the detailed report! Issue found, working on a fix.

urish, 2024-06-25:
Fix is up - can you please test and report?

SergioGasquez, 2024-06-25:
> Fix is up - can you please test and report?

The project is now working as expected and CI is back to green (https://github.com/esp-rs/no_std-training/actions/runs/9658595904/job/26655121912). Thanks for the quick fix!

SergioGasquez, 2024-07-03:
Hi! Just discovered that the issue still persists for Rust std, see https://wokwi.com/projects/333374799393849940?build-cache=disable

urish, 2024-07-03:
Thanks for reporting! So in nostd it works, but in std it doesn't, right?

urish, 2024-07-03:
Thanks! Issue found, working on a fix.

urish, 2024-07-03:
Fix is up. Looks like it was related to the previous issue - which partially fixed the problem (was triggering the interrupt correctly, but did not update the PCPU_INT register correctly on esp32s2 and newer chips). I hope that this time the fix is complete.

closed 2024-07-03 by urish

SergioGasquez, 2024-07-03:
Can confirm that the Wokwi project is working again, thanks for the quick fix!

urish, 2024-07-07:
Good news! We now also have automated regression testing for GPIO (including interrupts) across the supported chips:

https://github.com/wokwi/esp32-test-binaries/actions/runs/9824980776/job/27124560555