wokwi/wokwi-features #816: Wokwi for VS Code - Can't edit diagram need upgrade license?

antwal, 2024-06-29:
**Describe the bug**
I installed the extension, created a free license, restarted VS Code, created the diagram.json file and tried to modify it, it told me to upgrade the license 

**To Reproduce**
it doesn't allow you to modify the diagram, I have to deactivate the extension, modify it manually by creating the diagram from the website, save it and then reactivate the extension

**Expected behavior**
shouldn't it allow you to edit it directly from VS Code without going to the site?

**Environment (please complete the following information):**
 - OS: MacOS
 - Browser Chrome
 - Version 125.0.6422.142

**Additional context**

`VS Code Diagram:`

[image]

`Web Diagram:`

[image]

jonhp, 2024-09-24:
I am having the same/similar problem:
Windows 11
Edge Browser v129.0.2792.52
Wokwi v2.5.3
VSCode v1.93.1
I could not create a diagram; so I borrowed one from the web.
It's there, but I cannot change it; edit says upgrade and 'A' does nothing.
There is no "+" at the top of the editor panel.
I cannot click and drag a part.

djedu28, 2024-11-14:
I was unable to reproduce the error. I'm using the latest version.

Try updating to version 2.5.4

Link do Marketplace do VS: https://marketplace.visualstudio.com/items?itemName=Wokwi.wokwi-vscode

> Changelog
> 2.5.4 - 2024-10-14
> Bugfix: "Start Simulator and Wait for Debugger" does not indicate that gdbServerPort was not configured in wokwi.toml (https://github.com/wokwi/wokwi-features/issues/898)

moolight-seashell, 2024-11-14:
already the case : 
[image]
[image]

with os : manjaro

pekkanikander, 2025-01-14:
As far as I understand, **this is intentional,** not a bug. At the licensing page [https://wokwi.com/pricing](https://wokwi.com/pricing), the small i on the white circle after the "Wokwi for VS Code" feature of the Hobby+ plan includes a note on this.

So, AFAICS, the only bug here is that the error message is not clear enough. It should clearly say that one must upgrade to the Hobby+ or WokWi Pro plan. Of course, also the documentation could be more clear about this.

The error message can be found around here: https://wokwi.com/_next/static/chunks/pages/vscode/wcode-6eb4e9ead4cb6eda.js#L52 

( <rant> I hate debugging closed source obfuscated JS code, which the WokWi proper appears to be. Fortunately it is still JS, so such debugging is not completely impossible, just hard. </rant> )

urish, 2025-02-17:
Thanks for all the feedback, I've made it more clear now:

[image]

And also added documentation: https://docs.wokwi.com/vscode/diagram-editor

closed 2025-02-17 by urish