wokwi/wokwi-features #817: LCD 20x4 Works with Reversed Polarity

SailorMike, 2024-06-29:
**Describe the bug**
A clear and concise description of what the bug is. 
I mistakenly reversed the +5 V and Ground (pins 1 & 2) on the 20x4 LCD display and the display continued to function in the simulator

**To Reproduce**
Steps to reproduce the behavior, including link to a simulation project where we can see the problem.
I've corrected the problem in my simulation:
https://wokwi.com/projects/393639874390118401_

**Expected behavior**
The display should not work with Vss and Vdd reversed

**Environment (please complete the following information):**
 - OS: [e.g. iOS/Android/Windows]
 - Windows 7
 - Browser [e.g. chrome, safari]
 - Chrome
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

DarwinWasWrong, 2024-07-11:
The system isn't designed for voltage and current - you can remove the power lines and the item/screen will still function.

SailorMike, 2024-07-11:
Thanks, you're right. What confused me was that LEDs won't work when wired backwards.