wokwi/wokwi-features #820: Page dedicated to Club Members

iam-webdev, 2024-07-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

--->> Hello! I am hoping to see a page that displays a lot of featured projects and may include the works of the admins.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

--->>A page dedicated to club members where only club members can see more featured projects, so there will be a chance to improve the work of others. This will attract more people to join the Club. Thanks

drf5n, 2025-03-24:
There's the opposite:  a page deidicated to listing club mebers/makers:

https://wokwi.com/makers

djedu28, 2025-03-24:
> There's the opposite: a page deidicated to listing club mebers/makers:
> 
> https://wokwi.com/makers

Perfectly @drf5n. And on this page we can access the members' profiles and see each one's public projects.

djedu28, 2025-03-24:
For security reasons, no one can edit another user's project, but it is possible to create a clone of the project and edit it through it.

If anyone wants support, help with the project or to keep it open for community participation, it is desirable to create the project here on GitHub. 

And you can put information on how to enter/help the project in a `README.md` file created in wokwi, and it will appear in a special tab.

[image]