wokwi/wokwi-features #827: ESP32 GPIO34-39 INPUT NOT WORKING

kdirarslan, 2024-07-07:
Pins 34-39 of Esp32 are not used as input.  I connected the button and also connected the Pull-up and pull-down resistors, but nothing changes.  

I used these pins in the wokwi simulator before, but they became unusable a few weeks ago.

This is a very strange situation.

https://wokwi.com/projects/402769537997757441

urish, 2024-07-07:
Thanks for the detailed report! Uploaded a fix, can you please try again?

kdirarslan, 2024-07-07:
thank you problem solved

closed 2024-07-07 by kdirarslan

urish, 2024-07-07:
Excellent! The issue happened when we added the PCNT peripheral three weeks ago, as we implemented the FUN_IE bit of the IOMUX RTC. Previously, the IO MUX had the wrong pin count configured on the ESP32, but since FUN_IE wasn't implemented, it didn't cause any issue. Now I fixed the pin count, so the behavior should be correct.

kdirarslan, 2024-07-07:
I've been having problems with these pins for a while. I hesitated to write about the problem. I didn't think you would solve it in such a short time.
Thank you for your interest.