wokwi/wokwi-features #840: Improve efuse support on ESP32

PBrunot, 2024-07-08:
** Issue I have **

I have experimented Arduino 3.0.2 framework on ESP32S2 and ESP32S3 and the boot process fails early in efuse functions. It seems some changes in the bootloader (ESP-IDF, or Arduino, or some interactions of the two) are reducing compatibility with Wokwi.

You can reproduce the issue on a very simple PlatformIO + Arduino project here : https://github.com/PBrunot/esp32-ntp-clock/tree/esp32s3-test

Since the program hangs in esp_hal_read with infinite loop, and when forced out through GDB, hangs in efuse_hal_program, the efuse compatibility layer is probably part of the problem.

** Solution I'd like **

Improvement on the efuse simulation layer in Wokwi to solve the compatibility issue.

I open this issue request after discussion on Discord, see https://discord.com/channels/787627282663211009/1040739086400495677/1259505530335133696 for more details.

urish, 2024-07-09:
Thanks for the detailed report! 

Pushed an initial implementation of EFUSE write - can you please try again and report back?

PBrunot, 2024-07-09:
I did a quick test with the firmware on https://github.com/PBrunot/esp32-ntp-clock/actions/runs/9826190829 with https://wokwi.com/projects/new/esp32-s3
Before, it hung on boot before arduino setup()... Now the program reaches run() loop !
I'll test it more with a more complex program, but it really looks good so far! 
Thanks for the great support.

urish, 2024-07-09:
Thanks for testing!

Meanwhile, I also added regression testing for the efuse implementation, using the esp-idf efuse driver tests:

https://github.com/wokwi/esp32-test-binaries/actions/runs/9857718733

[image]

closed 2024-07-09 by urish