wokwi/wokwi-features #841: ESP32-S3 Arduino Core UART tests started failing

urish, 2024-07-09:
Last passing test was on July 2nd: https://github.com/espressif/arduino-esp32/actions/runs/9753984325
Tests started failing on July 3rd: https://github.com/espressif/arduino-esp32/actions/runs/9766805453

The following test is now failing: "periman_test"

```
2024-07-03 07:39:31 
2024-07-03 07:39:32 /home/runner/work/arduino-esp32/arduino-esp32/tests/validation/uart/uart.ino:108:periman_test:FAIL: Expected 'Hello from Serial1 (UART1) >>> to >>> Serial2 (UART2) while I2C is disabled' Was ''
```

urish, 2024-07-09:
Issue reproduced. There is a bug affecting UART2 reset on the ESP32-S3, causing it the peripheral to reset at the wrong time. Working on a fix.

P-R-O-C-H-Y, 2024-07-09:
@urish Working again. Thank you very much for fast fix.

closed 2024-07-09 by urish