wokwi/wokwi-features #843: esp32p4 and esp-idf 5.3-rc1 requires chip rev v0.1

petermm, 2024-07-12:
**Describe the bug**
Compiling project with esp-idf 5.3-rc1, leads to bootloader stopping on esp32p4:
```
2024-07-12 10:04:53 I (6) boot: chip revision: v0.0
2024-07-12 10:04:53 E (6) boot_comm: Image requires chip rev >= v0.1, but chip is v0.0
2024-07-12 10:06:33 API Error: Connection to wss://wokwi.com/api/ws/beta closed unexpectedly: code 1006
```

**To Reproduce**
compiling with official images https://hub.docker.com/r/espressif/idf/tags
Compile with 5.3-beta2 -> succeed.
5.3-rc1 -> boot_loader fail

**Expected behavior**
Works out of the box.

**Environment (please complete the following information):**
wokwi-ci etc.

**Additional context**
The default minimum is now rev1: https://github.com/espressif/esp-idf/commit/33c3d327c52bee87f43de5a0ef9707dee0a6d509

urish, 2024-07-12:
Pushed a fix - can you please try again?

petermm, 2024-07-12:
confirmed working, both beta2 and rc1 - https://github.com/petermm/AtomVM/actions/runs/9910866275/job/27382270623.

thanks!

closed 2024-07-12 by petermm