wokwi/wokwi-features #846: The ESP32-S3 with MicroPython does not have Serial output.

Koepel, 2024-07-19:
It is mentioned in the Discord channel that the combination of the ESP32-S3 with MicroPython causes troubles.
Starting with a new project does not show the Serial output: https://wokwi.com/projects/new/micropython-esp32-s3

urish, 2024-07-19:
Quick investigation shows it's probably related to the USB peripheral. Looking into this.

urish, 2024-07-20:
Further investigation shows that the RXFLVI USB interrupt is not correctly cleared, and keeps firing forever. Unfortunately, this is not a trivial issue to investigate why the interrupt is not cleared, as the USB OTG peripheral is very sparsely documented in the TRM, so it'll take some time until we figure it out.

Koepel, 2024-07-20:
If someone is looking for this Issue, is there a workaround ? Can one of the Arduino Serial ports be routed to the Serial Monitor ?

urish, 2024-07-20:
As a workaround, we can simply disable the USB on ESP32-S3 for all projects. 

The UARTs are already routed to the serial monitor - the issue is, if the USB peripheral is present, the code hangs on the initialization.

urish, 2024-07-20:
Following your feedback, I decided to disable the USB-ORG peripheral on the ESP32-S3 across all projects, so now micropython should be functional again.

closed 2024-07-20 by urish