wokwi/wokwi-features #850: Dependencies in the file "library.properties" might not be solved correctly.

Koepel, 2024-07-27:
This problem was mentioned in the Discord channel.
It will probably be solved in the library soon, but there might be a underlying problem with the dependencies.
Library: https://github.com/forntoh/LcdMenu

The LcdMenu has `depends=LiquidCrystal_I2C,LiquidCrystal` in the file "library.properties".
There is also a include in the library file "LcdMenu.h":
```
#ifndef USE_STANDARD_LCD
#include <LiquidCrystal_I2C.h>
#else
#include <LiquidCrystal.h>
#endif
```
It compiles in the Arduino IDE 1.8.19 in linux, but it does not compile in Wokwi.
Wokwi says: _"Build failed!_", "_Error installing LcdMenu: No valid dependencies solution found: dependency 'LiquidCrystal_I2C' is not available_".

It is a known Issue: https://github.com/forntoh/LcdMenu/issues/168

Test project: https://wokwi.com/projects/404566441698707457

The problem with this specific library might be solved soon. The maker of the library has already a pull request to separate the Menu from the LCD display library.

RoneFRANCE, 2024-09-17:
If this is the case regarding "underscore versus space", it would then be to request a library name change here : https://www.arduinolibraries.info/libraries/liquid-crystal-i2-c and dowload again the library on wokwi
Do you know how to do that ?
Thank you

Koepel, 2024-09-17:
I get the same dependencies problem with the Library Manager of Arduino IDE 2 with version 4.1.0 of the LcdMenu library.
It turns out that I did not read solution for the (closed) [Issue](https://github.com/forntoh/LcdMenu/issues/168). When the library is manually installed via a zip file, then it works. When the Library Manager is used, then there is a dependency problem for everyone.

Since the Wokwi Library Manager uses the same list as the Arduino IDE Library Manager, it behaves the same. 
Manually uploading a library in Wokwi (for paid users) is the solution, or else every source file from the library can be uploaded into the project (for non-paying Wokwi users).