wokwi/wokwi-features #851: Error loading library index

kraa965, 2024-07-28:
hello, when I try to add a library on the site wokwi.com I get the error message: Library index loading error:-(

urish, 2024-07-28:
Hello!

What do you see when you go to the following link? https://libraries.wokwi.workers.dev/list

(this is where the library index is loaded from)

kraa965, 2024-07-28:
> Hello!
> 
> What do you see when you go to the following link? https://libraries.wokwi.workers.dev/list
> 
> (this is where the library index is loaded from)

nothing the page is not loading

urish, 2024-07-28:
Then there is probably something that is blocking this page on your browser or network (e.g. anti virus, firewall, VPN, etc)

kraa965, 2024-07-28:
> Then there is probably something that is blocking this page on your browser or network (e.g. anti virus, firewall, VPN, etc)

thank you, I figured it out, for some reason the provider blocked this link https://libraries.wokwi.workers.dev/list

closed 2024-07-29 by urish