wokwi/wokwi-features #853: GDB in VS Code: ERROR: Unable to start debugging. Failed to find thread 1 for break event

mathias-buder, 2024-07-31:
**Describe the bug**
I'm trying to use GDB in VS Code. I followed the instructions on https://docs.wokwi.com/de-DE/vscode/debugging but I always get the error message `ERROR: Unable to start debugging. Failed to find thread 1 for break event`. I've attached the GDB extended log below for more details.

VS-Code Launch Configuration:

```
{
    "name": "Wokwi GDB",
    "type": "cppdbg",
    "request": "launch",
    "program": "${workspaceFolder}/.pio/build/uno/firmware.elf",
    "cwd": "${workspaceFolder}",
    "MIMode": "gdb",
    "miDebuggerPath": "${userHome}/.platformio/packages/toolchain-atmelavr/bin/avr-gdb.exe",
    "miDebuggerServerAddress": "localhost:3333",
    "logging": {
        "engineLogging": true
    }
}
```

**To Reproduce**
1. Compile a "Hello World" app using PlatformIO
2. Run "Wokwi: Start Simulator and Wait for Debugger"
3. Start the GDB debug session

**Expected behavior**
GDB should attach/start without issues

**Environment (please complete the following information):**
- Windows 10
- PlatoformIO (Core 6.1.15 , Home: 3.4.4)
- Wokwi Simulator Plugin 2.5.1

[gdb_log.txt](https://github.com/user-attachments/files/16447303/gdb_log.txt)

Weltschmerz-dev, 2025-01-26:
Same issue here. Did you every end up getting this to work? @mathias-buder

mathias-buder, 2025-01-27:
@Weltschmerz-dev, yes, I did resolve the issue. It was caused by `avr-gdb.exe` located in the directory `C:\Users\<user>\.platformio\packages\toolchain-atmelavr\bin`, not by `wokwi` itself. I downloaded the latest version of `avr-gdb.exe` and replaced the existing file. Although I don't exactly remember where I found the download, I've uploaded my complete VS Code setup, including `avr-gdb.exe`, for you [here](https://github.com/mathias-buder/Double-Door-Control/tree/main/tools/dev-env).

[image]

Let me know if that works for you too!

closed 2025-01-27 by mathias-buder

sgiraz, 2025-04-08:
> Although I don't exactly remember where I found the download, I've uploaded my complete VS Code setup, including `avr-gdb.exe`, for you [here](https://github.com/mathias-buder/Double-Door-Control/tree/main/tools/dev-env).

Hi @mathias-buder,

Maybe you downloaded it from https://github.com/ZakKemble/avr-gcc-build/releases

Weltschmerz-dev, 2025-04-14:
Totally forgot to leave a reply here. thanks for the fix @mathias-buder !