wokwi/wokwi-features #856: TCP connection to emulator device never disconnect

baiomys, 2024-08-05:
**Describe the bug**

Telnet or nc to emulated in VSCode device opens connection, but it remains open after disconnect from remote console.

```
wikwi.toml
[wokwi]
version = 1
firmware = 'build/flasher_args.json'
elf = 'build/main.elf'

[[net.forward]]
from = "0.0.0.0:1080"
to = "target:1080"
```

**To Reproduce**
I am not a member of club to run device connections in Web version, only VSCode

**Expected behavior**
Connection should close

**Environment (please complete the following information):**
 - OS: Windows and Debian
 - Browser VSCode
 - Version 1.92.0

**Additional context**
- VSCode IDE on windows, development server on Debian, esp32c3, Arduino under IDF project 
- Wokwi emulator 2.5.1

urish, 2025-02-07:
I have tested it in the simulator, and it seems like the simulator is indeed sending the FIN TCP packet upon client disconnect, in order to close the remote connection (10.13.37.1 is the simulator's IoT gateway, 10.13.37.2 is the simulated ESP32 device):

[image]

Not sure why the provided code doesn't detect it. Have you had a chance to test the code on a physical ESP32 device? Does it behave differently?

KlausPopp, 2025-02-07:
I can confirm that it works on a real device (ESP32S3). The server detects the connection as closed when the client terminates. But with woki, I never reach the "Connection closed" log output. 

```
    while (1) {
        pkt_hdr hdr = {0};
        len = recv(h->socket, &hdr, sizeof(pkt_hdr), 0);
        if (len < 0) {
            ESP_LOGW(TAG, "Error occurred during receiving: errno %d", errno);
            break;
        } else if (len < sizeof(pkt_hdr)) {
            ESP_LOGW(TAG, "Connection closed");
            break;
        }
   }
```

urish, 2025-02-11:
Thanks for explaining! Can you please share a complete code example  that I'll be able to compile and debug?

KlausPopp, 2025-02-24:
I have prepared an example here: https://wokwi.com/projects/423801095105925121.
But: Using the browser wokwi and the private IoT gateway, it works fine.
When I run the same in Vscode, I have the above mentioned problem

urish, 2025-03-02:
Thanks, issue reproduced!

urish, 2025-03-02:
Fixed and released version 2.6.5 of Wokwi for VS Code with the fix.

closed 2025-03-02 by urish