wokwi/wokwi-features #868: Chips C API Provides little to no error information

dakotath, 2024-08-25:
**Describe the bug**
I'm trying to use the I2C API for custom chips, and I'm getting errors as detailed as
```c
    .scl = scl,
           ^~~
1 error generated.

Error: Process exited with 1
```
**To Reproduce**
Try using the I2C API for custom chips by following the documentation.

**Expected behavior**
I expected a detailed error.

**Environment (please complete the following information):**
 - OS: Windows 11 Enterprise
 - Browser: Microsoft Edge
 - Version: 127.0.2651.86 (Official build) (64-bit)

**Additional context**
Me want to make custom chips.
[image]

urish, 2024-08-25:
Thanks! Can you please provide a link to the project above with the bogus error message?

urish, 2024-08-25:
Error messages fixed - should be more helpful now.

closed 2024-08-25 by urish