wokwi/wokwi-features #869: ESP32-C6 can't scan wifi when upgrading to IDF v5.3

lucasssvaz, 2024-08-27:
**Describe the bug**

In we are preparing to move to IDF 5.3. Only one test fails that is Wi-Fi scanning on ESP32-C6.
On hardware it works as expected. cc @P-R-O-C-H-Y

**To Reproduce**

https://github.com/espressif/arduino-esp32/actions/runs/10580718147/job/29319749870#step:11:591

**Expected behavior**
`WOKWI-GUEST` should appear in the network list as expected.

**Environment (please complete the following information):**
 - OS: ubuntu-latest

urish, 2024-08-27:
Thanks for reporting! Looking into this.

lucasssvaz, 2024-08-27:
@urish Forgot to mention that on IDF v5.1 everything works fine

urish, 2024-08-27:
Issue found, working on a fix.

urish, 2024-08-27:
Pushed a fix - can you please test again now?

lucasssvaz, 2024-08-27:
Worked as expected. Thanks!

closed 2024-08-27 by lucasssvaz