wokwi/wokwi-features #873: error: 'Serial0' does not name a type

step135, 2024-08-30:
```
In file included from /esp32/hardware/esp32/3.0.1/cores/esp32/Arduino.h:200,
                 from /sketch/ghgh.cpp:1:
/esp32/hardware/esp32/3.0.1/cores/esp32/HardwareSerial.h:356:16: error: 'Serial0' does not name a type; did you mean 'Serial'?
  356 | #define Serial Serial0
      |                ^~~~~~~
/sketch/ghgh.cpp:2:1: note: in expansion of macro 'Serial'
    2 | Serial.println("dddd");
      | ^~~~~~
```

urish, 2024-08-30:
Hello, this is not a simulation issue, rather an issue with the source code of the program. You can ask for help on the [Wokwi discord server](https://wokwi.com/discord), and don't forget to include the link to the project.

closed 2024-08-30 by urish

step135, 2024-08-31:
Why should I need any help?!!!