wokwi/wokwi-features #874: Downloading files from the simulated SD card to the computer

wzab, 2024-08-31:
**Is your feature request related to a problem? Please describe.**
When testing designs producing files on the emulated SD card, it is necessary to download them to the computer and analyze them here. Unfortunately, such functionality seems to be unavailable.

**Describe the solution you'd like**
It should be possible to download the selected file from the simulated SD card.

eMPee584, 2024-09-03:
Well I think you could do something like this:

```
#include <WebServer.h>
…
WebServer server(80);
…
SD.begin();
server.on("/download", serve_download);
server.begin();
…
void serve_download() {
…
  server.streamFile(file, "application/octet-stream");
…
}
```

urish, 2026-10-02:
This is now supported in both Wokwi for VS Code and the Wokwi CLI, using the new `writeback` setting: https://docs.wokwi.com/vscode/project-config#sd-card-contents

closed 2026-10-02 by urish