wokwi/wokwi-features #878: Support for big firmware

chconil, 2024-09-08:
When trying to build (or flash) a big firmware (I'm doing a star tracking system), even if the flash size seems to be correctly set, and the partition table too, following message appears.
"Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error during build: text section exceeds available space in board"

Minimal project: https://wokwi.com/projects/408486598166968321

On wokwi Visual studio and wokwi-cli, no errors appear but the emulated device keeps rebooting after the bootloader enters (making firmware smaller again solves it).

chconil, 2024-09-08:
Probable duplicate of https://github.com/wokwi/wokwi-features/issues/849, but the example focuses on the issue and makes it easy it reproduce / test.

chconil, 2024-11-14:
Up ?

urish, 2024-12-01:
Thanks for reporting! 

The flash size setting currently affects the simulation itself, not the compiler settings (so the compiler thinks the flash size is still smaller). I'll look if this is something we can easily fix.

urish, 2026-02-11:
For larger firmware sizes, they are not going to be supported in the web version compiler. Compiling the firmware yourself (and then simulating it either on the web, in VS Code, or the CLI) is the way to go.