wokwi/wokwi-features #880: VS Code Extension - possibility run Simulator and Debugger as separate Window

georgik, 2024-09-13:
Currently, the Simulator and Debugger run as new tabs in VS Code.

This is inconvenient because after starting a simulation and hitting a breakpoint, the Simulator disappears and stops. The user needs to reopen the simulator tab to resume the simulation.

One option is to pin the simulator to the right, but this reduces the visible code area.

A better option, especially for users with multiple monitors, is to drag the tab to the second monitor and let the simulator run there continuously. However, after stopping or restarting the simulation, the tab closes and a new one is created in VS Code, not as a separate window. The user must drag and drop the tab again to the second monitor.

Desired behavior: The Simulator should remain open as a separate window on the monitor where it was previously running.

djedu28, 2024-09-14:
Interesting feature. I don't have two monitors, but I always move the wokwi simulator tab to a new window. This keeps the simulation running while I open another code tab.

To avoid the window being closed, I press the restart button on the tab that opens with `diagram.json`.

urish, 2024-09-16:
Thanks for reporting! Should be fixed in the latest release (2.5.3) of the VS Code extension. 

Can you please test and report the results?

closed 2024-09-26 by urish