wokwi/wokwi-features #887: 14 Segment Display

Ali34-rgb, 2024-09-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add the 14 Segment Display, same features of the 7 segment but more readable, text can be printed only in this 14 segment display. But it have new functions.

djedu28, 2024-10-05:
There is already a 2-digit display (14 segments) on the wokwi.
[image]
https://wokwi.com/projects/410864745532814337

It is a variation of the 7-segment display. 
[image]

I recommend reading the wokwi documentation about `wokwi-7segment`
https://docs.wokwi.com/parts/wokwi-7segment

[image]

urish, 2025-01-10:
Unclear request, closing.

closed 2025-01-10 by urish