wokwi/wokwi-features #888: ESP-IDF JSON Support - Custom Naming

martysweet, 2024-09-24:
**Describe the bug**
With the VSCode plugin, I am looking to use the flasher_args.json functionality to read in a custom binary (containing NVS flash data).

AFAIK `flasher_args.json` is an internal idf framework file, and additional parameters cannot be easily added via a simple `idf.py build` command (I may be wrong, but haven't found much/any documentation on this).

To avoid conflicts, my workaround is to create a manual file called `wokwi_flasher_args.json` which will then:
- On build, copy `./wokwi_flasher_args.json` to `./build/wokwi_flasher_args.json`
- `firmware: 'build/wokwi_flasher_args.json'` is set

However, it looks like the current implementation may be checking for the exact path. 
The same file, with a different filename (within build) results in a simulation boot loop.
It would be ideal if it could see it as a .json, and derive that it is a flasher_args from the extension - or is this a known issue with the functionality?

**To Reproduce**
ESP-IDF project:
```
[wokwi]
version = 1
firmware = 'build/flasher_args.json'
elf = 'build/example_app.elf'
```

Works, however:

```
[wokwi]
version = 1
firmware = 'build/my_custom_flasher_args.json'
elf = 'build/example_app.elf'
```
Does not work.

**Expected behavior**
I would like custom naming support for flasher_args.json, to avoid any internal conflicts with idf.py tooling.

**Environment (please complete the following information):**
 - OS: Linux 
 - VSCode
 - Wokwi Extension v2.5.3

urish, 2024-09-24:
Hi Marty, you can achieve this by putting your own flasher_args.json in a different directory, e.g. `build/temp/flasher_args.json`. Wokwi looks at the file name (but not the complete path) to determine if it is an ESP IDF flasher file.

martysweet, 2024-10-03:
Hi @urish, thanks for the tip, that has worked nicely!

I ended up putting it in `./wokwi/flasher_args.json`.

martysweet, 2024-10-03:
Hi @urish,

It appears the change detection/autoreload logic isn't applied when the `flasher_args.json` file is not directly in `build/`.

When in `./wokwi/flasher_args.json` or `./build/wokwi/flasher_args.json` the entire "Wokwi Simulator" tab has to be closed and reopened to load the new binary (Stop/Start or Restart does not).

When in `./build/flasher_args.json`, the binary change is detected and the simulation automatically restarts.

Let me know if you need further details,
Thanks!

Marty

urish, 2024-10-14:
Interesting, will investigate