wokwi/wokwi-features #889: M5Stack Core S3 simulation crash when enabling display from ESP-BSP

georgik, 2024-09-26:
[ESP-BSP](https://github.com/espressif/esp-bsp) is standard a mechanism how to work with a board.

The initialization sequence in case of M5Stack CoreS3 includes also Power Up of display using AW9523.

The application that works on real HW, crashes in the simulation.

Here are artifacts: https://github.com/georgik/esp32-sdl3-example/actions/runs/11049438543

Here's the failure: https://github.com/georgik/esp32-sdl3-example/actions/runs/11049583033/job/30695464460#step:4:102
```
ESP-IDF video init
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4208ea11
file: "./managed_components/espressif__m5stack_core_s3_noglib/m5stack_core_s3.c" line 354
func: bsp_display_new
expression: bsp_enable_feature(BSP_FEATURE_LCD)

abort() was called at PC 0x4037d243 on core 0
```

I guess that this line backfires: https://github.com/espressif/esp-bsp/blob/master/bsp/m5stack_core_s3/m5stack_core_s3.c#L95
```
static uint8_t aw9523_P1 = 0b10100000;
....
    case BSP_FEATURE_LCD:
        /* Enable LCD */
        aw9523_P1 |= (1 << 1);
        break;
```

georgik, 2024-09-26:
Workaround: register two I2C devices with 0x58, 0x34
https://github.com/georgik/esp32-sdl3-example/commit/ca93f0247f9c26c4abb1f96ac7d53c5cc6bd1809#diff-cc1c20c7e52d5bb7960cef76ddbef85d4e0520ec4ddec9677de6a135ff83be73R6

urish, 2024-09-26:
I loved the creative use of the LCDs as mock I2C devices. Very clever!

closed 2026-01-15 by urish