wokwi/wokwi-features #890: FT6336 chip touch support - e.g. for M5Stack Core S3

georgik, 2024-09-26:
M5Stack Core S3 is using FT6336 for touch control.
Please, consider adding this chip, so that M5Stack Core S3 simulation could react on touch event, same like there is ESP32-S3-BOX-3 simulation with working touch.

urish, 2024-09-27:
We added initial FT6336 support - can you please test again and report?

georgik, 2024-10-01:
@urish Interestingly HW is working with also with FT5x driver which is in BSP, they seems to be somehow compatible with FT6: https://github.com/espressif/esp-bsp/blob/master/bsp/m5stack_core_s3/idf_component.yml#L14
While simulation seems not to be compatible with FT5x.
Here's declaration from esp_lcd_touch_ft5x06
```c
#define ESP_LCD_TOUCH_IO_I2C_FT5x06_ADDRESS (0x38)

/**
 * @brief Touch IO configuration structure
 *
 */
#define ESP_LCD_TOUCH_IO_I2C_FT5x06_CONFIG()           \
    {                                       \
        .dev_addr = ESP_LCD_TOUCH_IO_I2C_FT5x06_ADDRESS, \
        .control_phase_bytes = 1,           \
        .dc_bit_offset = 0,                 \
        .lcd_cmd_bits = 8,                  \
        .flags =                            \
        {                                   \
            .disable_control_phase = 1,     \
        }                                   \
    }
    ```
    
    This seems to be mistake on BSP level, yet it's interesting that it works with FT5x driver.

closed 2024-12-26 by urish