wokwi/wokwi-features #891: ESP32-S2 | Rust | no-std | Blinky not blinking

SergioGasquez, 2024-09-27:
**Describe the bug**
The Rust blinky example of S2 for no-std Rust is not blinking. If I download the elf and run it on real hw, I see the gpio changing as expected, but on the simulation, the led is always turned on.

**To Reproduce**
Build and run https://wokwi.com/projects/341955297469792851

**Expected behavior**
The led blinks as in real hw:
[image]

**Environment (please complete the following information):**
 - OS: Arch Linux (`6.10.10-arch1-1`)
 - Browser: Brave-Chrome

urish, 2024-09-27:
Thanks for reporting! Looks like the code hangs in `delay.delay_millis()` for some reason.

urish, 2024-09-27:
Pushed a fix - can you please test again?

SergioGasquez, 2024-09-29:
It's now working! Thanks for the fix!

closed 2024-09-29 by SergioGasquez