wokwi/wokwi-features #893: ESP-IDF examples in the documentation do no longer compile.

Koepel, 2024-09-29:
The examples in the documentation do no longer compile: https://docs.wokwi.com/guides/esp32#esp-idf-examples
They can run the pre-compiled binary, but they do no longer compile. The problem is: "_multiple definition of app_main_".

The first three have a ino file and no attribute `"builder": "esp-idf"`.
The fourth example has a main.c file, but no attribute either.

urish, 2024-09-29:
Thanks for reporting! Probably a regression introduced in the Arduino core for ESP32 v3.0.0. 

The proper solution would be to migrate these examples to use the "esp-idf" builder

OsvaldoTCF, 2024-10-01:
The examples were created when the C++ codes worked for IDF esp.

When we make the change specifying this, in the new standard, adding " { "builder": "esp-idf" }", the example breaks!

urish, 2024-10-01:
It wasn't C++ working with the "esp-idf" builder, as the examples used the standard Arduino builder.

urish, 2024-10-14:
Examples migrated!

closed 2024-10-14 by urish