wokwi/wokwi-features #897: Add FastLED.js compiler to your toolkit

zackees, 2024-10-12:
Hi there, I've merged a feature into FastLED:master but haven't announced this yet. But you will be most impacted so I wanted to give you guys some lead time.

FastLED now has a js compiler in docker. Communication between FastLED.js and the browser is still very alpha. But the proof of concept is there.

[image]

zackees, 2024-10-12:
Attached is the wasm build of examples/wasm after compilation

[fastled_js.zip](https://github.com/user-attachments/files/17348215/fastled_js.zip)

urish, 2024-10-14:
Congrats on merging the feature! Not sure how is Wokwi impacted by this, can you please explain?

zackees, 2024-10-14:
You have a paid service for emulating boards. A chunk of your users use it for fastled. Before FastLED.js, this was the only solution for running FastLED in the cloud.

Now FastLED will be cloud native.

I wanted to do the due diligence of letting you know first. But it seems like the impact isn't that great so good to know.

urish, 2024-10-15:
Thanks for explaining. It's great to read that FastLED will be cloud native!

closed 2024-10-15 by urish