wokwi/wokwi-features #898: Wokwi: Start Simulator and Wait for Debugger does not indicate that gdbServerPort was not configured in wokwi.toml

georgik, 2024-10-14:
**Describe the bug**
When developer selects Start Simulator and Wait for Debugger and the `gdbServerPort` was not configured in wokwi.toml. There is no indication that the configuration is missing and Debugger just times out.

**To Reproduce**
- create `wokwi.toml` without `gdbServerPort=3333`
- VS Code: Start Simulator and Wait for Debugger
- VS Code: Start debugging
- result: timeout

**Expected behavior**
The Simulator should display warning and sugegestion how to correct it. E.g. by adding `gdbServerPort=3333` to `wokwi.toml`

urish, 2024-10-14:
Excellent suggestion, thanks! Released this as part of version 2.5.4 of the extension.

closed 2024-10-14 by urish

georgik, 2024-10-21:
Works very good. Thank you.