wokwi/wokwi-features #899: wokwi emulator responds to the rfc2217 connections only when vs code is in the foreground

petrarce, 2024-10-19:
I am using rpi pico w configured setup for running the wokwi emulator in VS code  based on the mycropython firmware 1.23.0. However, i would like to be able to connect to repl and also perform the filesystem operations with the running emulator using the `mpremote` tool. This is right now problematic, since the serial communication with the emulator over the rfc2217 tcp port is not working properly, unless the VS code plugin is moved on the foreground. More specifically:
1. I connect to the emulator with mpremote using this command:
```bash
mpremote connect rfc2217://localhosta:4000 repl
```
2. Typing anything to the console doesn't produce any echo from the serial port
3. Switching to the VS code seems to flush the serial communication buffers.

Within the wire-shark i can see this picture:
[image]
It seems to me, that when the VS code is not on the foreground the wokwi simulator is somehow paused. 

Is it an expected behavior? Is it possible to somehow configure the wokwi VS code plugin to run on the background instead of pausing when the VS code is not on the foreground?

djedu28, 2024-10-21:
> Is it an expected behavior?

Yes, it is expected behavior.

> Is it possible to somehow configure the wokwi VS code plugin to run on the background instead of pausing when the VS code is not on the foreground?

As a hack I highlight the wokwi tab in VSCode, this way it remains “active” even though it is not in the foreground.
The automatic translation may not be great, so I leave a GIF of the process.

https://github.com/user-attachments/assets/d8834e10-03a9-4e98-b522-174917419722

urish, 2024-10-28:
> Is it an expected behavior? Is it possible to somehow configure the wokwi VS code plugin to run on the background instead of pausing when the VS code is not on the foreground?

Yes, this is the expected behavior - simulation is usually CPU intensive, and it would be very frustrating to find out you ran out of laptop battery because you forgot it running in the background. That being said, it's a good idea to have a way to tell Wokwi to keep running even while the tab in hidden, for use cases like yours.