wokwi/wokwi-features #900: Firefox times out at 30 seconds or gives a message that it can not fet a resource.

Koepel, 2024-10-22:
It was mentioned in the Discord channel that the Firefox browser stops the build process after 30 seconds.
At that time it was not consistent for me, but now it is. Every ESP32 (or variant) with Arduino times out at 30 seconds, but sometimes at 60 seconds.

Today, sometimes this error shows after 30 seconds:
[image]

The build process is normal when using the Chrome browser. It tries to build up to 100 seconds.

urish, 2024-10-22:
Thanks for reporting - I reproduced with [the following project](https://wokwi.com/projects/321525495180034642).

urish, 2025-01-10:
Does this still happen? I tried reproducing it today but it didn't reproduce for me.

Koepel, 2025-01-13:
No problems so far.
I think this Issue can be closed.

closed 2025-01-13 by urish

Koepel, 2025-01-18:
Update: the problem is still there.

I have not seen it with ESP32 projects.
Sometimes with Arduino Uno and Mega projects, the Firefox browser times out at about 16 or 38 seconds.

urish, 2025-01-19:
Interestingly, it just happened to me with Chrome:

[image]

The ~16 seconds mark seems very arbitrary...

urish, 2025-01-19:
Luckily, I was with the devtools open, and it seems like the actual request took longer than 16 seconds:

[image]

I'm not yet sure at which point the dialog popped up. Hopefully it'll reproduce again and I'll be able to collect more data.

Koepel, 2025-01-27:
I wrote before: "_I have not seen it with ESP32 projects_", but it happens also with ESP32 projects.