wokwi/wokwi-features #902: Wokwi VSCode Extension not working with Platformio

ZanzyTHEbar, 2024-10-27:
**Describe the bug**
The Wokwi vscode extension is failing to execute any code that i am building. I have used the simplest examples, and nothing happens.

**To Reproduce**
`diagram.json`:
```json
{
    "version": 1,
    "author": "nomadspring",
    "editor": "wokwi",
    "parts": [
        {
            "type": "board-xiao-esp32-c3",
            "id": "esp",
            "top": 0,
            "left": 0,
            "attrs": {}
        }
    ],
    "connections": [
        [
            "esp:TX",
            "$serialMonitor:RX",
            "",
            []
        ],
        [
            "esp:RX",
            "$serialMonitor:TX",
            "",
            []
        ]
    ],
    "dependencies": {}
}
```

```toml
[wokwi]
version = 1
elf = ".pio/build/esp32c3_debug/firmware.elf"
firmware = ".pio/build/esp32c3_debug/firmware.bin"
rfc2217ServerPort = 4000
[[net.forward]]
from = "localhost:8180"
to = "target:80"
```

`main.cpp`:
```cpp
#include <Arduino.h>

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);
    delay(100);
    Serial.println("Hello, ESP32-C3!");
}

void loop() {
    // put your main code here, to run repeatedly:
    delay(10);  // this speeds up the simulation
}
```

**Expected behavior**

- start simulator
- firmware is executed

**Environment (please complete the following information):**
 - OS: Windows and Linux
 - Browser: VSCode
 - Version: latest

**Additional context**
The simulator opens and runs, says that it is simulating - however the output is blank. Nothing. No LEDS light up, no serial is printed, nothing happens. I also get nothing over the serial TCP interface.

urish, 2024-10-27:
Hello, can you try changing diagram.json as follows?

```json
{
    "version": 1,
    "author": "nomadspring",
    "editor": "wokwi",
    "parts": [
        {
            "type": "board-xiao-esp32-c3",
            "id": "esp",
            "top": 0,
            "left": 0,
            "attrs": {}
        }
    ],
    "connections": [ ],
    "dependencies": {}
}
```

ZanzyTHEbar, 2024-10-27:
Yes, that seems to work. Removing the serial connections is necessary to get output in the vscode extension? My circuits and such seem to work now.

urish, 2024-10-27:
Great!

It's necessary for the XIAO series, which uses the USB for serial output by default

closed 2024-10-27 by urish

ZanzyTHEbar, 2024-10-27:
> Great!
> 
> It's necessary for the XIAO series, which uses the USB for serial output by default

I had the same issue on all esp32c3 boards that i tried, including the `board-esp32-c3-devkitm-1` default. Been trying to debug it for hours xD

urish, 2024-10-28:
That must have been frustrating. Glad you opened the issue to ask!

It actually depends on how the Serial port in the board is configured - even the XIAO can be configured to write to the hardware UART. However, for the XIAO, USB is the default. Perhaps USB is also the default for all C3 based boards in [PlatformIO](https://platformio.org/).