wokwi/wokwi-features #904: Feature request: Keyboard shortcuts for start/stop/pause in the diagram

Koepel, 2024-10-29:
It was mentioned in the Discord channel that there are no keyboard shortcuts in the diagram to start and stop and pause the simulation.

There are keyboard shortcuts for the editor: https://docs.wokwi.com/keyboard-shortcuts/
There are keyboard shortcuts for designing the diagram: https://docs.wokwi.com/guides/diagram-editor/#keyboard-shortcuts
But there are no keyboard shortcuts when the simulation is running and the diagram window is active.

The Ctrl+Enter to start would be nice, and also keyboard shortcuts to pause and stop.

urish, 2024-10-29:
Good call - any suggestions for what the keyboard shortcuts would be?

drf5n, 2024-10-29:
I'd think the command-Enter to start should match the command in the editor window.

And that it shouldn't conflict with the potential button hotkeys.  For example: 

[image]

Interesting: the arrow keys move the sketch around unless they are bound to buttons.

The SerialMonitor window appears to be a separate context

Maybe . and , for stop and pause would seem natural and low-impact?

| Description | Windows/Linux | Mac |
|--|--|--|
| Start simulation | Ctrl + Enter | ⌘ Enter |
| Stop simulation | . | . |
| Pause Simulation | , | , |

urish, 2024-10-31:
Thanks for the useful suggestion! 

I've pushed a version with these keyboard shortcuts - would love to get feedback (and if it works well, we can close the issue)

Koepel, 2024-10-31:
The . and the , work but I can't start it.

drf5n, 2024-10-31:
Command-Enter ',' and '.' work on a Mac in Chrome with https://wokwi.com/projects/384583073378554881

Cool.

urish, 2024-11-01:
Thanks @Koepel  - which browser do you use? I tested with Chrome / Firefox

Koepel, 2024-11-01:
**Tested** with focus on different fields with different browsers and different settings:
- Start: with Ctrl+Enter
- Pauze: with ',' and Ctl+,
- Stop: with '.' and Ctrl+.

Browsers:
- Windows 11: Microsoft Edge, Firefox, Vivaldi, Chrome
- linux: Firefox, Chromium

**Good results:**
It works as expected.
The Firefox browser has a "search for text when you start typing" option. That messes up the ',' and the '.' as expected. But the Ctrl+, and Ctrl+. work normally.
The Vivaldi browser had always some compatibility issues with Wokwi, but it seems to work better now. The keyboard shortcuts do always work.

**One thing:**
I can not replicate the result where ',' and '.' work and Ctrl+Enter not.
But I can set the focus wrong.
When the focus is on the source code and then I click on the "Simulation" tab or the bar next to it, then the circuit field does not get the focus.
[image]

urish, 2024-11-03:
Thanks! Then I guess we can close this issue as complete. I'll look into the focus thing as well.

closed 2024-11-03 by urish