wokwi/wokwi-features #906: Autoformat in custom chip.c files

drf5n, 2024-11-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I'm always frustrated when I try to edit a custom chip .c file and the autoformat doesn't work.  (I'm using Chrome on a Mac and Shift-Option-F simply inserts an Ï  )  Also, the other-click menu doesn't show formatting as an option:

[image]

Here's a sample project with the issue:

https://wokwi.com/projects/413585628218621953

I would like the custom chip editor window on Wokwi to be able to autoformat custom chip.c files like it does for other .ino files and .json files.

There's a bit of discussion at:

https://discord.com/channels/787627282663211009/1302645970294673460/1302645970294673460

I think  the .chip.c suffix confuses something.

urish, 2024-11-04:
Thanks for reporting! Pushed a fix - can you please test again now and report?

drf5n, 2024-11-04:
It works for me.  Thanks!