wokwi/wokwi-features #907: Mfrc522 Rfid Reader please!!

AbdennourBELKACEM, 2024-11-06:
we need the Mfrc522 Rfid Reader module in the wokwi simulator please!

djedu28, 2024-11-07:
label > duplicate #359 and #643

## Hello, I'm a member of the WOKWI community and I'll help as much as I can.

Hello! We understand your anxiety regarding the implementation of Mfrc522. The team is already aware of this request and it's being tracked in issue #359.

Repeatedly sending requests does not speed up the process. Please use the main issue #359 instead.

The implementation of Mfrc522 depends on various factors such as priority (number of votes), as well as the complexity of integration and community demand.

The Mfrc522 feature was opened for voting on May 20, 2022. However, the feature still lacks sufficient votes to be prioritized. The more votes a feature receives, the higher the chance of it being implemented. And so far, it has only acquired 12 votes, so it doesn't have priority for implementation. https://wokwi.com/features#feature-1242782423

[image]

## If Mfrc522 is essential for your project, we suggest some alternatives while waiting:

1 **Physical testing:** Purchase the component and perform traditional tests.

2 **Custom development:** Explore the creation of a custom chip using the [`Wokwi Custom Chips`](https://docs.wokwi.com/chips-api/getting-started/) tool. This option requires a deeper understanding of how Mfrc522 works and the hardware description language used by Wokwi.
> In #643, I presented an alternative that I developed for using the CI Mifare RC522 module

3 **Community:** Contact other members of the WOKWI community. Perhaps someone has already developed a similar solution or can assist with implementation.

4 **WOKWI Community Projects:** Research WOKWI community projects that have implemented Mfrc522 using `Wokwi Custom Chips`.

We appreciate your understanding and hope this information is useful.

> Automatically translated. 
-----
Original message (pt-BR):

>## Olá, sou membro da comunidade-WOKWI e ajudo no que posso.
>
>Olá! Entendemos sua ansiedade em relação à implementação do Mfrc522. A equipe já está ciente dessa solicitação e ela está sendo acompanhada na issue #359.
>
>O envio repetido de solicitações não agiliza o processo. Por favor, utilize a issue principal #359.
>
>A implementação do Mfrc522 depende de diversos fatores, como prioridade (quantidade de votos), assim como a complexidade da integração e a demanda da comunidade.
>
>O recurso Mfrc522 foi aberto para votação em May 20, 2022.  No entanto, o recurso ainda não possui votos suficientes para ser priorizado. Quanto mais votos um recurso recebe, maior a chance de ele ser implementado. E até o momento só adquiriu 12 votos, logo não tem prioridade para ser implementado. https://wokwi.com/features#feature-1242782423
>
>## Se o Mfrc522 é essencial para seu projeto, sugerimos algumas alternativas enquanto isso:
>
>1 **Teste físico:** Adquira o componente e realize os testes de forma tradicional.
>
>2 **Desenvolvimento customizado:** Explore a criação de um chip customizado utilizando a ferramenta [`Wokwi Custom Chips`](https://docs.wokwi.com/chips-api/getting-started/). Essa opção exige um conhecimento mais aprofundado sobre o funcionamento do Mfrc522 e sobre a linguagem de descrição de hardware utilizada pelo Wokwi.
>> em #643 apresentei uma alternativa que desenvolvi para usar o modulo `CI Mifare RC522`
>
>3 **Comunidade:** Entre em contato com outros membros da comunidade Wokwi. Talvez alguém já tenha desenvolvido uma solução similar ou possa te auxiliar na implementação.
>
>4 **Projetos da comunidade wokwi** Pesquise por projetos feitos pela comunidade wokwi que tenha implementado o Mfrc522 utilizando `Wokwi Custom Chips` .
>
>
>Agradecemos sua compreensão e esperamos que essa informação seja útil.

closed 2024-11-07 by urish

anton21m, 2025-07-17:
I DID IT! 🎉

After 4 weeks of sleepless nights and deep dives into low-level emulation…
The RC522 chip is now fully simulated in Wokwi! 🧠⚡

✅ The MFRC522 Arduino library is now working right in the browser
✅ You can even connect multiple RFID readers with unique UIDs
✅ Works seamlessly with real Arduino code and examples

Try it yourself:

🔬 Full MFRC522 test project:
👉 https://wokwi.com/projects/432456562791073793

📚 Multiple readers with unique UIDs:
👉 https://wokwi.com/projects/436744920653539329

📦 Chip export demo:
👉 https://wokwi.com/projects/436746939600791553

Found a bug? Have ideas?
Create an issue here:
🐞 https://github.com/anton21m/wokwi-rc522-rfid-chip

Let’s build the future of virtual electronics together 🚀

urish, 2026-02-02:
Good news: the MRFC522 is now officially supported on Wokwi!

Docs: https://docs.wokwi.com/parts/board-mfrc522
Example project: https://wokwi.com/projects/454873133691473921