wokwi/wokwi-features #909: Hello requesting parts for ESP32: 1)Active Buzzer 2)Waterflow sensor 3)Water Valve 4)Soil Moisture Sensor 5) ESP32 Dev Kit V1

Hutgib81, 2024-11-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Hutgib81, 2024-11-07:
Pwease hweap mee

djedu28, 2024-11-07:
I recommend adjusting this issue.
Put your requests in the issue description. List and describe what each of the components are, and how you intend to use them.

In wokwi, the `parts` are independent of the microcontroller, so the Arduino `parts` work with `ESP`, `STM`, `Raspberry Pi Pico`, ...

djedu28, 2024-11-07:
Generally speaking, you can replace any analog sensor with a potentiometer and a digital actuator with an LED.

Therefore, you can exchange the active buzzer for an LED. Just like you can replace the *Soil Moisture Sensor* with a potentiometer

1) Active Buzzer => LED
3) Water Valve => LED
4) Soil Moisture Sensor => Potentiometer

5) ESP32 Dev Kit V1 => `board-esp32-devkit-c-v4`

The esp32 devkit v1 already exists on wokwi, however it has been discontinued. If possible, use [`esp32-devkit-c-v4`](https://wokwi.com/projects/new/esp32). It is compatible with the DEV kit V1. And in wokwi `dev kit-c-v4` is the default for new ESP32 projects.

If `ESP32 Dev Kit V1` is really necessary, put it in `diagram.json`
```json
...
"parts": [
    { "type": "board-esp32-devkit-v1", "id": "esp", "top": 0, "left": 0, "attrs": {} }
  ],
...
```
example: `diagram.json`
```json
{
  "version": 1,
  "author": "Anonymous maker",
  "editor": "wokwi",
  "parts": [
    { "type": "board-esp32-devkit-v1", "id": "esp", "top": 0, "left": 0, "attrs": {} }
  ],
  "connections": [ [ "esp:TX", "$serialMonitor:RX", "", [] ], [ "esp:RX", "$serialMonitor:TX", "", [] ] ],
  "dependencies": {}
}
```

djedu28, 2024-11-07:
These replacements are just to be able to simulate the project, and code without having to change the code to test on the physical component.

And by making some abstractions in the code, it is possible to exchange the `2)Waterflow sensor` for a Potentiometer

closed 2024-11-07 by urish

Hutgib81, 2024-11-07:
My project is "SMALL SCALE FARMING SOLUTION (CROP MONITORING & MANAGEMENT)"
I will attach the schematic I made from EasyEDA and the list of
components that I need (Not available in Wokwi).

List of components:
1) Water Flow Sensor
2) Active Buzzer
3) Water Valve
4) Soil Moisture Sensor

I am truly sorry for the inconvenience caused, I just need these
components for simulation purposes.