wokwi/wokwi-features #911: rfid module feature is demanded

AbdennourBELKACEM, 2024-11-07:
the iot tinkering community is looking  up for an rfid module because no common simulator have a rfid module for simulation (direct simulation),not tinkercad ,not simulIDE ,not even proteus ,if you add it to wokwi it will be a big advantage for you and for the community

djedu28, 2024-11-08:
I understand your enthusiasm for adding an RFID module to Wokwi.  However, creating multiple duplicate issues can be frustrating for you and especially for developers.

Let's focus on constructive dialogue:

1. **Review existing discussions**: Check if your idea has been previously suggested or discussed. Issue #359 might be relevant.

2. Contribute positively: Instead of opening new issues, consider:
   - Upvoting existing feature requests
   - Exploring current workarounds or alternatives
   - Study and detail the operation of the Mfrc522 Rfid, publishing in #359
   - Develop your own custom chip using the [Wokwi Custom Chips](https://docs.wokwi.com/chips-api/getting-started/) tool.

3. Implement a [mock function](https://medium.com/@rickhanlonii/understanding-jest-mocks-f0046c68e53c) RFID tag simulation
  a. Implement the `read_card()` function FAKE:
      - Use a conditional statement to check if the button is pressed
      - the button represents a tag that has been brought closer to the sensor
      - Return a predefined tag ID string ("ABCDEF..." in this example)
      - Consider adding randomness or dynamic generation of tag IDs
      
      With this approach, you can continue development, abstracting/isolating the RFID from the application

4. Implement a mock RFID tag simulation:

   a. Create a custom chip that mimics RFID tag behavior:
      - Implement a simple state machine to represent the tag's lifecycle (idle, detecting, authenticating, etc.)
      - Simulate the tag's response to reader queries
      - Include functions to update the tag's state and return simulated data

   b. Develop a mock RFID reader interface:
      - Create methods to send read/write commands to the tag
      - Implement error handling and timeout mechanisms
      - Simulate various tag responses based on distance, orientation, etc.

5. Utilize creative approaches to simulate RFID functionality using current resources:

   a. Mock communication: Simulate RFID card interactions by mocking the SPI communication protocol. You can create a custom chip that mimics the Mfrc522's behavior, allowing you to test RFID-like scenarios. [Wokwi Custom Chips](https://docs.wokwi.com/chips-api/getting-started/)

   b. State machine simulation: Implement a state machine that mimics the RFID reader's operation, including card detection, authentication, and data transmission.

   c. Hardware abstraction layer: Create a custom chip that abstracts the RFID functionality, allowing you to interact with it through a simplified API.

   d. Emulation-based testing: Develop a system that emulates RFID tags and readers, allowing you to test RFID-like scenarios without needing actual hardware.

   e. Protocol analysis: Study and replicate the RFID communication protocols used by real devices, implementing them in Wokwi using existing components.

6. If you have difficulty with the [API Wokwi Custom Chips](https://docs.wokwi.com/chips-api/getting-started/), consider using another simulator that is compatible with more than one microcontroller at the same time, and use an Arduino simulating the RFID module, using SPI communication.

----

Remember, while these alternatives may not perfectly replicate real-world RFID systems, they can help you develop and test RFID-related logic within the constraints of Wokwi's current capabilities.

## Remember, Wokwi is a free tool developed by volunteers. While your contributions are valuable, it's important to respect the time and effort put into its development.

Let's work together to make Wokwi an even better tool for all users.

<details>
<-- 
já disse para não spamar. o wokwi é uma ferramenta gratuita, você não pode exigir nada DO UNICO DESENVOLVEVOR que te forneceu essa otima aplicação sem cobrar nada. Tenha respeito pelo tempo investido por ele na aplicação, e pare de spamar, fazendo-o perder tempo.

Recomendo parar de importunar criando novas issues. Se tá incomodado com isso, ESTUDE E FAÇA VOCÊ MESMO USANDO A API DE CHIPS PERSONALIZADOS
-->
</details>

AbdennourBELKACEM, 2024-11-10:
sorry sir ,but i don't understand all this amount of anger in your details section of your replay , it was  just adding 2 issuses for the same feature, anyway i appreciate your  constructive dialogue part.

closed 2024-12-06 by urish