wokwi/wokwi-features #912: Input period symbol stops simulation

al-ilyin, 2024-11-08:
**Describe the bug**
When I print period symbol in the serial monitor input textbox simulation stops.
When comma symbol printed simulation pauses.

**To Reproduce**
Example project trying input float number.
https://wokwi.com/projects/413969335903682561

**Expected behavior**
Expected that string with float number parsed into variable.

**Environment (please complete the following information):**
 - OS: Windows 10 (64 bit)
 - Browser chrome
 - Version 130.0.6723.116
 - Browser firefox
 - Version 131.0.3

**Additional context**
Didn't find any description of that behavior in documentation, but some my projects need to input float numbers.

urish, 2024-11-08:
Fixed!

closed 2024-11-08 by urish

al-ilyin, 2024-11-08:
> Fixed!

Thanks a lot, it's great now!