wokwi/wokwi-features #915: wokwi-ci: not run rp2040 firmware

kassane, 2024-11-13:
Hi, @urish :wave: 

**Describe the bug**

wokwi-ci not run rp2040 `.uf2` firmware, emit error:
```console
API Error: Connection to wss://wokwi.com/api/ws/beta closed unexpectedly: code 1006
```

**To Reproduce**

- https://github.com/kassane/rp2040-dlang/compare/main...wokwi-ci

**Expected behavior**

No error! Run, like experimental/viewer
- https://github.com/kassane/rp2040-dlang/releases/tag/v1.0.0 (wokwi-testing)

**Environment (please complete the following information):**
 - OS: Linux
 - Browser: Firefox
 - Version: 132.0.1

## Reference
- https://github.com/kassane/rp2040-dlang/issues/1

urish, 2024-12-01:
Hi @kassane,

Thanks for reporting!

Can you please try using the [wokwi-ci-server-action](https://github.com/wokwi/wokwi-ci-server-action) and see if it fixes the issue?

kassane, 2024-12-02:
> Can you please try using the [wokwi-ci-server-action](https://github.com/wokwi/wokwi-ci-server-action) and see if it fixes the issue?

https://github.com/kassane/rp2040-dlang/actions/runs/12120844427/job/33790453083
```console
Run docker run -d -p 9177:3000 --name wokwi-ci-server wokwi/wokwi-ci-server
Unable to find image 'wokwi/wokwi-ci-server:latest' locally
latest: Pulling from wokwi/wokwi-ci-server
da9db072f522: Already exists
9f55c94602e4: Pulling fs layer
cd7875902382: Pulling fs layer
23b7f06a0560: Pulling fs layer
23b7f06a0560: Verifying Checksum
23b7f06a0560: Download complete
9f55c94602e4: Verifying Checksum
9f55c94602e4: Download complete
9f55c94602e4: Pull complete
cd7875902382: Verifying Checksum
cd7875902382: Download complete
cd7875902382: Pull complete
23b7f06a0560: Pull complete
Digest: sha256:eb4f70811fc63614fa4b2b077b7e2486aa21d7108ee2086a7d8f3ca060469a4a
Status: Downloaded newer image for wokwi/wokwi-ci-server:latest
cdd7682e3d56e8d786251d85abd53050790818b4a5d81f9a4f2cfa6ffbfb3c49

Run wokwi-cli --timeout "5000" --expect-text "$CI_EXPECT_TEXT" \
Wokwi CLI v0.14.5 (e69493bb8261)
Connected to Wokwi Simulation API 1.0.0-20241117-gdfc711df
Starting simulation...
Timeout: simulation did not finish in 5000ms
Hello, world from D!
Error: Process completed with exit code 42.
```

urish, 2024-12-02:
looks like it's running now - the only question is why it didn't match the expected text.

Is the "Hello, world from D!" print followed by a newline?

kassane, 2024-12-02:
Solved!!

closed 2024-12-02 by kassane