wokwi/wokwi-features #918: Simulation speed of ESP32-S3 in ESP-IDF mode drops below 10% after 4min28.

Koepel, 2024-11-16:
It was mentioned in the Discord channel that the ESP32-S3 drops the speed after 4 minutes and 28 seconds.
I have run tests for all the ESP32 variants in both Arduino and ESP-IDF mode.
My starting point was here: https://wokwi.com/projects/new
I forced a re-compile by changing the source code.

I can confirm that only the ESP32-S3 drops the speed to about 3% after 4 minutes 28 seconds and only in ESP-IDF mode. 
Tested with the Firefox and the Chromium browser.
The Rust no-std is okay for the ESP32-S3.
The Rust STD example does not have a endless loop.

urish, 2024-11-16:
Thanks for reporting! So the steps to reproduce would be to go to https://wokwi.com/projects/new/esp-idf-esp32-s3, start the simulation, and wait for 4 minutes and 28 seconds, right?

Koepel, 2024-11-16:
Right.

urish, 2024-11-16:
Reproduced!

urish, 2025-02-17:
Looks like there's an issue with the SysTimer peripheral implementation when the timer value goes over 32 bits. Investigating.

urish, 2025-02-17:
Fixed!

closed 2025-02-17 by urish