wokwi/wokwi-features #923: Please add FM Receiver

Ali34-rgb, 2024-12-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

djedu28, 2024-12-02:
Hi,

Improve the text of this `issue`, describe which component you want to be included in wokwi. 
It needs to be a commercial receiver. To be added to wokwi it needs to be commercial hardware and accessible by the community at large. Or else you will need to create a custom chip for your application.

Give examples of real use that uses this feature, so it will make it easier to develop the chip thinking about what you need.

## Example of questions that this text needs to describe: 

1. Is the device you want for communication between more than one Arduino? 
2. Or an 'FM Radio Receiver'?

For the former, there is `STX882RF` (but it works with **433Mhz**)
For the second, there is the `TEA5767` module (FM receiver and transmitter).

## You can build your own `FM Receiver`

    https://docs.wokwi.com/chips-api/getting-started

<details>
<summary>original text pt-BR
</summary>
Olá.

Melhore o texto dessa `issue`, descreva qual componente deseja a ser incluindo no wokwi. 
Precisa ser um receptor comercial. Para ser adicionado no wokwi precisa ser um hardware comercial e acessível pela comunidade em geral. Ou então você precisará criar um chip personalizado para tua aplicação.

Dê exemplos de uso real que utilize esse recurso, assim facilitará desenvolver o chip pensando no que precisa.

## Exemplo de perguntas que esse texto precisa descrever: 

1. O dispositivo que deseja é para comunicação entre mais de um Arduíno? 
2. Ou um `FM Radio Receiver`?

Para o primeiro, existe o STX882RF (mas ele funciona com 433Mhz)
Para o segundo, existe o modulo TEA5767 (receptor e transmissor FM).

## Você pode construir seu próprio `Receptor FM`

    https://docs.wokwi.com/chips-api/getting-started

</details>

djedu28, 2024-12-02:
If you don't have experience with the 'api chips' https://docs.wokwi.com/chips-api/getting-started
I recommend starting with something simpler. 

* example: #884 create the simulation of the 14-segment display with 'wokwi chips-api'

<details>
<summary>original text pt-BR
Se não tiver experiencia com a `chips-api` https://docs.wokwi.com/chips-api/getting-started
Recomendo começar com algo mais simples. 

* exemplo: #884 criar a simulação do display de 14 segmentos com `wokwi chips-api`
</details>

closed 2024-12-06 by urish