wokwi/wokwi-features #925: Anti Spam Measure in Latest projects

DarwinWasWrong, 2024-12-02:
The instances of spam in latest projects is increasing

It is noted that none of the spam entries actually can compile

Change the title **Latest projects** to  **Latest compiled projects**

1. do a syntax check or compile to test the project before adding to  Latest compiled projects
2. do not add projects that are created by new users - eg a few days old account.

This would also stop **Latest compiled projects** from having quick un-working examples.

urish, 2024-12-03:
Thanks for reporting! 

I've started working on measures to significantly reduce spam, and deleted over 250,000 existing spam projects from the database. Please do let me know if you come across any spam projects on Wokwi.

urish, 2024-12-08:
Still working on eliminating the spam projects, got 427,269 down so far.

It's challenging, as some of them are pretty sophisticated and actually look like legit projects, e.g. 

[image]

DarwinWasWrong, 2024-12-08:
Wow. 

Impressive spam kill tally 

 

but what lengths they go to.

I really didn't think they would make it able to compile.

 

Never underestimate 

 

 

  _____  

 

image.png (view on web) <https://github.com/user-attachments/assets/0a7e400c-7441-4ec4-a71e-6b5f75d3d49e>

urish, 2024-12-09:
Yeah, never underestimate!

Removed total of 471,001 SPAM projects created by 3,887 different registered users, and put a bunch of measures to stop / slow down future SPAM raids. Closing - please do open a new issue if you notice another SPAM raid in the future.

closed 2024-12-09 by urish